voroshkov/Chorus-RF-Laptimer #1: Android app > no main screen

TJohn7002, 2017-03-05:
After launch of Chronos app there's only white screen, No any keys/writings is shown.
I'm using Yotaphone 2 (android 5.0)

voroshkov, 2017-03-05:
no app and tab names as well?

TJohn7002, 2017-03-05:
You can find screenshot by the link:
https://drive.google.com/uc?export=download&id=0B3SQNU593aT5bVhJSEVNRk91ZVk
If I can provide smth else which can help - please say. Only one issue - my phone isn't rooted.

voroshkov, 2017-03-05:
Well, I'm stuck here... It works fine on all my android devices. And I'm not yet so experienced in Android development to easily troubleshoot, sorry.
Maybe try to enable bluetooth and pair your HC-06 before opening the app.

TJohn7002, 2017-03-05:
Yes, no any tab. I've paired it to phone already.
Also - I've checked it on my pretty old tablet - it is working on it. Tab is using android 4.1.1.
Possible issue on my phone is caused by screen resolution , it is fullhd

TJohn7002, 2017-03-07:
Closed - After restart of phone app is working fine.

closed 2017-03-07 by TJohn7002