voroshkov/Chorus-RF-Laptimer #2: [Feature Request] Add new screen in Android - BandScanner

TJohn7002, 2017-03-14:
Hi Andrey,

Is it possible to implement analogue of Band Scanner from sheaivey/rx5808-pro-diversity?
It would be pretty useful.
https://hackadaycom.files.wordpress.com/2016/01/bandscanner.jpg?w=800
[image]

voroshkov, 2017-03-14:
Yes, It was on my list by the way ;)

But I thought about implementing it as a separate application instead of loading the Chorus Laptimer interface, which is mostly aimed for racing. What do you think?

TJohn7002, 2017-03-14:
IMO, adding of 1 or 2 tab screens will not make App interface overloaded.

traced, 2017-03-22:
Including this into the app would be great!

ps915, 2017-06-17:
would be awesome

AABatteries, 2017-06-17:
@voroshkov +1 for the Band Scanner feature, it would be great to see it inside the app (as a separate tab like @TJohn7002 mentioned)