voroshkov/Chorus-RF-Laptimer #3: [Feature Request] Freq management

TJohn7002, 2017-03-14:
Hi Andrew,

I've got several ideas regarding frequencies management in Android app:

1. Display Freq, not only Channel and Band
2. Add ability to auto choose Using Freq from app\on device start-up (to make it more independent from android app):
2.1. Plug in battery in drone
2.2 Turn on Chorus device (or press it's button) (at ~1-2 m distance)
2.3 Chorus should scan all available bands and choose the one with strongest RSSI (~). Play sound when ready, go into Timer mode, play sound again
3. Save last used settings in App\or (and) Chorus device (Pilot name and Freq)

P.s. Timer mode is just like Race mode, but without limitations of lap number - just to save the timings

voroshkov, 2017-03-14:
Completely agree with 1 and 3. 
What concerns 2 - it seems useful for a single racer, but looks inconvenient when you prepare a competition of several drones.

TJohn7002, 2017-03-14:
I was thought about this.
But there's workaround I've found:
1) In case if you you have single-RX device - I completely agree with you, there can be issue. Needed "Next" button as it done in  rx5808-pro Autoscan feature
2) In case if you have multi-RX device (able to measure >1 freq at the same time), then auto-scan can work step-by-step:
1) Search first strong Freq, assign it to first RX module
2) Continue search for next bands, and assign it to next modules.

But I agree -there could be collisions of bands because some chinese Tx are transmitting at several channels at the same time =(

voroshkov, 2017-03-14:
That makes sense. From UI standpoint it can be as simple as adding a button on a "Freq" tab, like "Autoscan", which will automatically scan known bands and assign different signal peaks to different devices!
So you'll just have to turn on all drones and click "Autoscan" to have bands/channels assigned.

TJohn7002, 2017-03-14:
Yep, that's I was thinking about)
Let's device itself do it's work)
Moreover I think it is possible to make Chorus device standalone:
add OLED screen and 3 buttons to make it independent from android app... This will increase cost of project by ~4$ but make it completely different from other devices that are already in the market.

voroshkov, 2017-03-14:
Well, the latter is questionable :)
Adding OLED will require separate controller for it (one more Arduino) and won'be convenient for organizing competitions, which was the primary target for Chorus.

TJohn7002, 2017-03-14:
Ok, just  ignore it than)