voroshkov/Chorus-RF-Laptimer #4: [Feature Request] Android app > Use speech API to say last\best times

TJohn7002, 2017-03-14:
I think it will be great to hear last lap time once app got it from Chorus device.
Also it will be useful to hear best lap time right after last lap time.
smth like "Last lap: 23.6 sec... Best lap: 18.2 sec"
For the new best lap it will be just like "New Best lap: 15.8 sec"

highway11, 2017-03-14:
I agree this is a great idea. It  is a very beneficial feature to hear your lap times called out as you are flying with the goggles on!

Another open source lap timer I built does this. It runs in the web browser though, and I much prefer the Android app interface you have built for this one.

voroshkov, 2017-03-26:
https://github.com/voroshkov/Chorus-RF-Laptimer/pull/10

closed 2017-03-26 by voroshkov