voroshkov/Chorus-RF-Laptimer #5: Application doesn't seem to read BT data after reconnecting

voroshkov, 2017-03-15:
1. Power on Chorus device.
2. Connect from mobile app -> settings and current RSSI measurements are shown in the app
3. Disconnect 
4. Reconnect again -> RSSI values are not being updated. 
(not always reproducible)

FranzAr, 2017-03-15:
I can confirm this behaviour. If you kill the app completely and then restart it, it works again.

KingModest, 2017-03-17:
So there are actually two cases, when this bug appears:
1) launch app -> if connects -> exit with back button -> launch app -> connect -> (bug - not connecting properly)
2) launch app -> if connects -> disconnect -> connect -> (bug - not connecting properly)

The only way - is to kill the process of the app and connect

voroshkov, 2017-03-17:
Can you specify the symptoms of "not connecting properly?"

voroshkov, 2017-03-17:
As discussed: The app stops interaction with Bluetooth (doesn't transmit/receive data).
in the 1st case described by @KingModest FREQ and PILOTS tabs are empty.
in the 2nd case - UI remains, but buttons don't work (no data transmission occurs)
(Observed in Android 7.1.1)

commit ca455bd9 (2017-03-18): fix #5, fix #8 (Android app reconnect issues)
closed 2017-03-18 by voroshkov