voroshkov/Chorus-RF-Laptimer #9: Client application for different platforms.

Rikbruggink, 2017-03-18:
Hi!,

Would it be possible to create apps for different platforms like IOS or a as standard webapp for different embedded platforms like the Rasberry PI?

voroshkov, 2017-03-18:
Unfortunately I don't plan to learn iOS development in the nearest future. I'd be happy to assist if any iOS developer would join the initiative and create the app.

Rikbruggink, 2017-03-18:
Would it be possible to share the serial commands specification?

voroshkov, 2017-03-18:
It's already available in wiki: https://github.com/voroshkov/Chorus-RF-Laptimer/wiki

traced, 2017-03-20:
Just read about iPhones and the cheap HC-05 and HC-06 bluetooth modules. They don´t work with iPhones.

Rikbruggink, 2017-03-21:
Found the same info that the HC-* series are not compatibel with the Iphone, HM-10 is and around the same cost. @voroshkov thank you for the wiki link. I don't have the time currently to create an iphone app, sorry for that. Will follow the repository for updates.

nickspiel, 2017-06-14:
I am going to give a web app a crack. Just waiting on the parts to arrive. Should have something in a couple of weeks.

ps915, 2017-06-15:
awesome. looking forward to this

timofpv, 2017-10-10:
I could help with an WiFi webserver / homepage based soloution with an ESP8266. Would work with every device with an Browser and WiFi - laptop smartphone even with an ebook reader

voroshkov, 2017-10-10:
Hey, that's what I'm going to try in the nearest future. Would you contact me in facebook messenger to share/coordinate?

voroshkov, 2017-10-19:
App for iOS is in beta testing stage! You can join testing as described here: https://github.com/voroshkov/Chorus-RF-Laptimer/blob/master/README.md#hot-news-

nikybiasion, 2018-01-28:
Any news about a web app?

voroshkov, 2018-01-28:
nope. sorry. out of my priorities for now

Zai, 2019-04-21:
We are actually working on an integration of chorus to our web app : RotorMatch.com :)

voroshkov, 2019-04-22:
@Zai, let me know if you need any assistance regarding Chorus API or have any integration questions

Zai, 2019-04-22:
@voroshkov thx ! Actually I'm learning your API to see how I can wrap it to work with ours. ;)
I come back soon with a first integration to test. :)