voroshkov/Chorus-RF-Laptimer #12: RSSI not shown on last device

FranzAr, 2017-03-23:
Have 4 devices, on the Pilots tab the last device RSSI shows "1". If I do "Set" it works correctly. Laps are also triggered correctly.

[image]

[image]

voroshkov, 2017-03-23:
Did you update the Android app and Arduino firmware since last weekend?
Is the issue persistent? Does it work this way right after connection and after disconnect/connect cycle?

FranzAr, 2017-03-23:
Yes, both latest version

voroshkov, 2017-03-23:
Do you have the problem with 4 devices only? Can you describe the use case?

FranzAr, 2017-03-23:
Yes, had no problems like that with only 2 devices. Was just testing on the bench, but I could reproduce this behaviour every time. Gonna do some more investigation, when I have the time. On the Freq Tab it shows the right value, it's only on the Pilots tab. Just a small issue but wanted to let you know.

voroshkov, 2017-03-23:
Got it. Thanks. Will take a look.

voroshkov, 2017-03-24:
I guess I've found the reason of the described issue.
@FranzAr, can you try the updated APK from attached archive and let me know if it fixes the problem?

[ChorusRFLaptimer-release.zip](https://github.com/voroshkov/Chorus-RF-Laptimer/files/867188/ChorusRFLaptimer-release.zip)

FranzAr, 2017-03-24:
Great, I'll try it tonight

FranzAr, 2017-03-24:
Yes, problem seems to be solved, I get plausible values now on all 4 devices.

voroshkov, 2017-03-24:
Cool. Will include the fix into next release then.

commit e3c52eca (2017-03-25): fix #12 (no RSSI displayed for last device)
closed 2017-03-25 by voroshkov