voroshkov/Chorus-RF-Laptimer #17: Idea: Alternative connection through FTDI/UART

Szpero, 2017-04-13:
Reason: BT and 2.4GHz are often jammed by frsky and other stuff. 433MHz for example will handle without a problem with much more longer distances.

Idea: Connect FTDI converter through OTG cable.
To FTDI you can connect any device like:
- straight cable to arduino
- 433MHz (or other) modems on frequencies which are not in use by race quads.

Application: need to be changed.

Arduino side: no need - just any other device than BT module.

Hardware: For example FTDI based on CH340G (tested with library) and pair of transmiting devices which can handle 115kbps on air.

Proposed library: https://github.com/mik3y/usb-serial-for-android

I'm not so expeirenced with java but for now only FTDI library is merged.

traced, 2017-04-13:
Can 433Mhz modules handle the baud rate?

Szpero, 2017-04-14:
Many 433MHz data links for pixhawk/apm telemetry are able to use 250kbps air speed to maintain 115kbps inner connection for device.
Here is example of device which is working at different frequency but its able to work at 250kbps also:
http://www.banggood.com/3DR-Radio-Telemetry-915MHZ-Module-For-APM-APM2-p-73539.html

From what I've seent most of those devices are using:
- USB with build in FTDI converter
- transmiting modules which can be configured with AT commands.

Within end of month I'm going to replace BT module (config link) on my Pixhawk flyer. It is good oportunity to check link device for other purposes :)

tigert, 2017-04-15:
Would it work with a ESP2866? Those are arduino compatible, so you could run the whole thing in one, and setup a wifi hotspot on your phone. 

You could even log the results in a mqtt server or talk to the app over wifi.

traced, 2017-04-15:
ESP8266 works with 3.3v, so you would need something to provide two voltages (3,3v for ESP and i think 5v for the RX module).

tigert, 2017-04-15:
If you have one of those modules with usb, like the nodemcu amica or adafruit huzzah, there is the usb/5V pin if you power it via usb. I do that to power a DHT11 temperature sensor that also needs 5V and it works.

tigert, 2017-04-15:
Or maybe some logic level converter might work.

closed 2017-07-01 by voroshkov