voroshkov/Chorus-RF-Laptimer #19: Android apk build

mclei, 2017-05-01:
I am not Android developer, so maybe stupid question, but I am not able to compile the application with this error:

FAILURE: Build failed with an exception.

* What went wrong:
A problem occurred configuring project ':app'.
> Could not resolve all dependencies for configuration ':app:_debugApk'.
   > Project :app declares a dependency from configuration 'compile' to configuration 'default' which is not declared in the descriptor for project :bluetoothspp.

Please could you help me?

voroshkov, 2017-05-01:
I use latest Android Studio for compiling. It should work without issues.

mclei, 2017-05-01:
I am also using latest Android Studio on Debian/Squeeze. Both Studio and "gradlew assemble" throws this error.

voroshkov, 2017-05-02:
Well, to be honest, I'm not an Android developer either. Hope somebody more experienced could help here.

gpolic, 2017-05-02:
That is a long shot, but are you sure you are using "Import Project."?
Usually when trying to open a project directly there are various problems. Import project, is taking care of the dependencies.

Also try to update Gradle and see if it makes a difference.

I am on windows and the latest Android Studio and I cannot recreate this error message.

mclei, 2017-05-02:
I am also not Android developer. I am using "Import Project". Have gradle 3.5 and Android Studio 2.3.1.
Why I want to compile the project is that there is explicitly forced English for Text to Speech and I want to use the system settings.

gpolic, 2017-05-03:
I am not Android Developer either, but I am trying to learn.

Looks like the Text to Speech code is hard coded for U.S. settings
I dont know how to change that but will have a look

voroshkov, 2017-05-03:
Yes, it's all hardcoded so far. In order to Internationalize we'll need to make localization for the entire app. (all texts, user-defined spoken messages, etc.)

voroshkov, 2017-05-03:
Btw, the code becomes more and more ugly with adding new features (due to flaws in initial design and lack of Android development experience). Need to perform thorough refactoring before implementing anything significant.

mclei-asw, 2017-05-03:
But your project is COOL and we are already using it in the field :-) Thanks.

mclei, 2017-05-03:
But your project is COOL and we are already using it in the field :-) Thanks.

gpolic, 2017-05-04:
Inside TextSpeaker.java in the onInit method
You can use this to obtain the default locale, and pass it to setLanguage instead of "Locale.US"

@TargetApi(Build.VERSION_CODES.N)
public Locale getCurrentLocale(){
    if (Build.VERSION.SDK_INT >= Build.VERSION_CODES.N){
        return getResources().getConfiguration().getLocales().get(0);
    } else{
        //noinspection deprecation
        return getResources().getConfiguration().locale;
    }
}

I found this info here:
http://stackoverflow.com/questions/14389349/android-get-current-locale-not-default

Also this is very useful:
http://stackoverflow.com/questions/38189386/texttospeech-setlanguage-not-working

mclei, 2017-05-04:
Yes, I have ideas how to change it, but I am not able to compile the project and this task is about it :-)

gpolic, 2017-05-04:
I would do the change in a PR, but I am unable to test :-(  
maybe @voroshkov can compile that for you and include it in the project

As for the refactoring. Yes it maybe needed specially if new functionality is being added. You could start by doing adding more test units, to assist in more efficient refactoring.

voroshkov, 2017-05-04:
As for including this into the project:
Guys, I don't think using system locale is enough for the app to work properly. All messages should also be localized and auto-selected upon choosing a locale. 
Also this may result in not speaking pilot names (if they don't correspond to system locale): e.g. I tried using my name in Russian, but Speech Engine with US locale just doesn't pronounce it.

gpolic, 2017-05-04:
oops! yeap you're right. Translation is needed.

lnx13, 2017-06-19:
`Project :app declares a dependency from configuration 'compile' to configuration 'default' which is not declared in the descriptor for project :bluetoothspp.`
This error occured becose you use case sensitive file system, just rename bluetoothSPP to bluetoothspp

closed 2017-06-19 by voroshkov

mclei, 2017-06-19:
Confirmed, never noticed it is that simple :-)

PR #27 (2017-06-19): Fix build issue on case sensitive file system
resolve #19