voroshkov/Chorus-RF-Laptimer #20: Help with Arduino Nano

androidiano78, 2017-05-06:
Hi,
I just finished assembling a Solo device, but with Arduino Nano (not having Arduino Pro Mini). For now I have missed the part for voltage Lipo monitoring. That said, everything connects to the application via the bluetooth module, but I did not understand, the FREQ android app page, what is that number next to Device # 1, in my case 234. Going to the PILOTS page, RSSI Threshold gives me Always 1, and until I insert the above mentioned number (234), but it does not give me the option to set it, START RACE (1PILOT) does not appear on the RACE page. Once START RACE starts and I try to make a ride with the quadcopter (walking naturally for now, both with powered and non armed motors, but I do not think it makes any difference) when I step in front of the device nothing happens and does not ring buzzer. I would need help to figure out what the problem is, especially with wiring, since I am using Arduino Nano, but it should be compatible. I have only one device, so I did not connect the 4 pins to series more devices; In addition I did not even have the bridge to shorten two pins of Bluetooth (loopback jumper in scheme), I seemed to understand that it was to do for multiple connected devices.

My PinOut and configuration:

I supply everything from lipo - bec USB to mini USB - Arduino nano;
Ch1-Ch2-Ch3 (RX5808) - 1k resistors - D12-D11-D10 (Arduino nano);
GND (RX5808) - GND (HC-06 Module) - one of the 2 GNDs (Arduino nano);
+ 5v (RX5808) - 5V (Arduino nano) - VCC (HC-06 Module);
RSSI (RX5808) - A6 (Arduino nano);
Positive (Buzzer) - D6 (Arduino nano);
GND (Buzzer) - one of the 2 GNDs (Arduino nano);
RX0 (Arduino nano) - TXD (HC-06 Module);
TX1 (Arduino nano) - RXD (HC-06 Module).

I attach some photos and thank you in advance.
[image]
[image]
[image]
[image]
[image]

voroshkov, 2017-05-06:
Why do you connect RSSI to A6?

androidiano78, 2017-05-07:
Oh my god, I apologize for the lapse and for the time that I made you lose .... I put the pinout, which among other things I was wrong also the CH, and now magically everything works. Now I have to test it only in flight. Congratulations on the work done

voroshkov, 2017-05-07:
Good to have it solved! :)

closed 2017-05-07 by voroshkov