voroshkov/Chorus-RF-Laptimer #22: Built with 5 SOLO issue

vannoz, 2017-05-27:
Hello, I built a Chorus version with 5 SOLO. I have an issue when selecting players: name of 5th and 1st are the same and 5th SET rssi value doesn't change from 0, rtx receive the signal, but button SET doesn t work..only strange bip...any idea?
(Wirimg is ok, Uno pro are all ok...)
thanks

voroshkov, 2017-05-30:
Can you make sure you use the latest app? Does it have LiPo monitor adjustment feature?
If yes, then please specify what Android device you are using.

voroshkov, 2017-05-31:
Is Freq tab entirely visible on the screen or the 5th item is off?

vannoz, 2017-06-01:
Hello Andrey, here you can see a video from my mobile:
https://youtu.be/iBP88XJ0LvI
Thanks
Ciao
Andrea

voroshkov, 2017-06-01:
Thanks for the vid. I'll try to investigate what's happening. Looks like incorrect handling of  lists in android app. I'd appreciate if experienced Android devs could join and help here.

vannoz, 2017-06-01:
Yes...unfortunately i'm olny Android user :)
maybe depends on the software you have used to build the Android app...I
tried the MIT App Inventor...but is not easy

maddust, 2017-06-21:
I'm having the same problem . Any update with this issue?

vannoz, 2017-06-21:
I think nothing up to now...Andrey is working to solve

voroshkov, 2017-06-22:
Does anybody experience the same issue having different number of nodes? 4, or 6?

vannoz, 2017-06-22:
With 4 units no problems...i don't have a 6th unit to do the test sorry

vannoz, 2017-07-04:
ciao Andrey, I have some suggestios.
I tested this Sunday the laptimer chorus RF...mine is 5 unit...but if you
remember only 5 are working on the android app...
I used with 4 racers per time...we selected only 4 VTX frequencies...and
all the time I was changing the pilot name...then loosing a lot of time
My request is> is it possible to add a page where to list all players name,
then when preparing the partecipants to a race, select the names of the
participants from a drop down menu?
it will be also a good things to assign a COLOR code to some frequency...
it will be also a good things to see the MHz freq value per each
channel/band...because some time the VTX on the quad has not the same table
MHz vs channel/band...

thanks
ciao
Andrea

voroshkov, 2017-07-04:
@vannoz valid points, but out of scope of this bug. Would you create a feature request(s) instead?

highway11, 2017-07-04:
I'm also experiencing this bug. I recently added 2 more modules to upgrade from a 4 unit to 6 unit device. In the Pilots tab any changes I make to the 6th module get mirrored to the first one. I also cannot disable the 6th module to continue with 5 as the checkbox does not seem to work on the 6th as well

closed 2017-07-04 by voroshkov

vannoz, 2017-07-04:
> via 7a952cb

Thanks. So I just have to install the new version?

vannoz, 2017-12-14:
have you seen this?
https://www.banggood.com/LANTIAN-5_8G-Wide-Voltage-Bluetooth-Timer-Personal-Practice-Version-for-FPV-Racing-Drone-p-1237993.html?rmmds=search&cur_warehouse=CN

voroshkov, 2017-12-14:
Yep. Already discussed in FB group :)

vannoz, 2017-12-14:
Ah ok...sorry :)

voroshkov, 2017-12-14:
Oh, you really shouldn't be! Thanks for notifying anyway!

vannoz, 2017-12-14:
You did a great job...so I was scared seen others gain money with your idea!

voroshkov, 2017-12-14:
If I could make huge money of it I'd try to :) So far I'm happy as it gains popularity as long as the license permits :)

commit 7a952cb6 (2017-07-04): fix #22 (inability to setup nodes higher than 4)