voroshkov/Chorus-RF-Laptimer #23: Only one Pro Mini

mario123456789o, 2017-05-31:
Hi,
wouldn't it be possible to use one Arduino only. Wire all the spi pins as in the rx5808 diversity, but use different pins for the chip Select pins. So if you have for rx5808, wire chipselect 1 to d1, cs 2 to d2 ...
So you would use only one arduino, the stackability is a bit worse, because you have to limit it to number of rx5808, but to be honest who needs more than 8 different modules?
If my rx5808s ever arrive, I will test this.
Greetings 
Mario

voroshkov, 2017-06-01:
Theoretically it's possible, but out of scope of this project, sorry.

mario123456789o, 2017-06-01:
Ok
I'll try my idea as soon as my rx5808s arrive, and if it works I'll notify you :).
The limitation is that we only have 8 analog pins available(at least for my arduino :D) so a maximum of 7 rx5808s can be used. But this should be enough for most people otherwise you have to sacrify the battery pin.
The other problem might be the performance, that the arduino is too slow to read 7 analog inputs in realtime

closed 2017-07-01 by voroshkov