voroshkov/Chorus-RF-Laptimer #25: Bluethooth Range

ps915, 2017-06-06:
Hey Guys,

Whats your Bluethooth Range? HC05 has a pretty bad range so here is an idea.

To keep the Tracker Compact, we might use a class 1 bluethooth module. That will get us 100+ meters range: https://www.sparkfun.com/products/12580

The bad thing is, that most smartphones dont support class 1 except the iPhone 7 i think.

So, maybe its possible to write an Chrome App, or standalone App for a Laptop and use this Class 1 Bluethooth Adapter.

https://www.sparkfun.com/products/9434

A friend also tested various wifi modules like ESP... he only got 10 meters and had TCP/IP packages losts

...or..

is there any "Android App" Emulator which accepts bluethooth connectivity?

ps915, 2017-06-06:
I have an other idea for increasing the range dramatically without loosing the bluethooth feature. 
Lets build a master slave class 1 bluethooth gateway. Should work:
https://cdn.sparkfun.com/datasheets/Wireless/Bluetooth/bluetooth_cr_UG-v1.0r.pdf

[image]

highway11, 2017-06-06:
I find it makes a big difference if you can elevate the bluetooth module above the ground. I use CAT5 cable to extend mine out and away from the Chorus box.  I can get around 50m away from it and still have communication that way. Its elevated maybe 1.5 meters off the ground.

ps915, 2017-06-06:
Thanks for your contribution. 50m with Cat5 is awesome. i should try that. If it works stable, the above solution is overkill, indeed! =)

highway11, 2017-06-06:
My CAT5 cable is only about 4 meters long. Then I just mount it on a stick to elevate it :) Very low tech solution, but it works for me!

ps915, 2017-06-25:
I just bought this as a range enxtender =) 

Chorus -> UART RF  .>.>.><.<.<. UART RF -> HC05/Laptop (Gui from Daniel Müller​)

https://www.aliexpress.com/item/2x-UART-RF-Wireless-Transceiver-Module-USB-to-TTL-RS232-Telemetry-Kit-2-4G-for-Arduino/32239562366.html?spm=2114.13010108.99999999.10.ITKskN

FranzAr, 2017-06-25:
Phil, does ist work like expected?

ps915, 2017-06-25:
I'll report back asap I'll hold it in my hands. ;)

FranzAr, 2017-06-25:
Ah, thought you got it already. I'll be waiting ;-)

rcdesigns, 2017-07-15:
I hang my timer from a tree branch and easily reach 20m to where we sit. I'm sure it would go further.

rcdesigns, 2017-07-15:
I presume soldering on a better antenna would improve things a lot