voroshkov/Chorus-RF-Laptimer #28: 8in1 PCB Version 2

ps915, 2017-06-25:
OK guys,

a while ago, i created a 8in1 PCB for the Chorus Tracker. It was my first design and of course the PCB was not 100% correct. There were some GND Pins of the Arduino not connected together "GND Island Fail". 

Now i came up with the Idea to create a stackable design. 

So there will be a 10x10 PCB with all the RX5808 receiver on top. You can either direct solder them or stackable with Pin Headers.

A second PCB, the Main PCB will be connected also with pin headers. 

I will create a second Main PCB for the Delta 5 Tracker as well. So one can choose if he wants to build delta5 or chorus.... or even another tracker.

All Arduinos from the PCB before can be used for this PCB as well. 2 pin headers were added becaus i included a buzzer which is shared with all 8 nodes via some schottky diodes. 

There is also an option to mount two XT30 connectors to the PCB and also a stepdown (Pololu 5V, 2.5A Step-Down D24V25F5) to power the whole unit. 

Why two XT30? If the battery will run out of power, just connect a fresh one and charge the old one. 

An other feature is the Rj45 jack where you can plug in a network cable to extend the range of your bluethooth.

What is your opinion? Whats missing? 

[image]
[image]
[image]
[image]
[image]
[image]
[image]

Rikbruggink, 2017-06-26:
It looks really great Phillip!. Are you gonna release the pcb's to seedstudio or osh?

ps915, 2017-06-26:
sure, they will be released on Seeedstudio when the testing is done and everything is fine

ps915, 2017-07-10:
Update: Just decided to go with 8in1 PCB, no stackable RX PCB :-P

Ok guys. I spend a few weeks designing Chorus 8in1 V2 which should be the final version.
Features: 
- 10x10 PCB
- up 8 Nodes (if you use 6 nodes, just bridge TX/RX)
- 1 shared buzzer for all nodes (protected by diodes)
- 2x XT60 Input
- 2x MicroUSB 6V Input (Adafruit MicroUSB Breakout Board)
- 1x Serial Output RJ45 Socket
- 2x Serial Output Pin-Header
- 5V Stepdown ready (MP1584EN)
- all parts stackable (parts can be replaced easily, also better airflow, less heat)

As always, i will test it before publishing, but i think i wont find any issues this time =)
After that, you can order it yourself on seeedstudio.com
[image]
[image]
[image]
[image]
[image]
[image]

cbnlnk121, 2017-09-04:
@ps915 Maybe I'm missing it, but what's up with the dual XT60? I assume is runs right to the 5V stepdown first? Sorry for the newb question, just having a hard seeing the trace.

cbnlnk121, 2017-09-05:
Deleted my prior comment ... Re-reviewing the trace, it's dual inputs. Great idea! Thanks!

Chewpasta, 2017-12-03:
hi, i have a cuestion in this project, do you have a pcb with only 4 nodes?  ty!

ps915, 2018-02-16:
I wont release the 8in1. Instead i nearly finished the 4in1. Stay tuned!

closed 2018-02-16 by ps915