voroshkov/Chorus-RF-Laptimer #29: Feature Request: WS2812 LEDs

ps915, 2017-06-26:
Why not hooking up a WS2812 LED Stripe to the first node?

1. Pilot Colors
Every Pilot gets a color and the start gate will shine in the pilots color when passing it. 

2. Visual Countdown
add a bright ws2812 led panel in front of the starting pads or even the whole start gate.
So there is a visual start sign. When the race is finished or the pilot has max laps, the gate will show red. 

What do you think? 

Code: https://github.com/adafruit/Adafruit_NeoPixel

FrufoFPV, 2017-06-26:
I really like the idea. Its pretty much the same as TBS has implemented for their Racetracker and SmartGates.

Ziege-One, 2017-06-26:
Nice
 I would build it as a new module like the others. 
All information comes over the serial line.
So you pick them up

AABatteries, 2017-06-28:
I would recommend using the FastLED library as opposed to NeoPixel one - it's more versatile, and offers better performance than the latter. 

It would be a great addition, I might make a pull request once I have it working (but no earlier than a month time). I think adding a connector (just 3 pins) to make it possible to plug in a cable to both power (from the same LiPo) and send signal to the LEDs, so that the tracker can be placed away from the gate, would be the easiest/best way to physically implement this.

Repo for FastLED can be found here: [FastLED library on GitHub](https://github.com/FastLED/FastLED)

reptil1990, 2017-07-10:
I started coding! Could need some help: https://github.com/reptil1990/Chorus-RF-Laptimer

voroshkov, 2017-07-12:
Idea for possible implementation of this feature: instead of loading the Solo module with new features you may like to create a separate Arduino module which will just proxy all commands as other modules do, but implement only its specific functionality (drive LEDs in our case). So the chain of modules might look like:
Solo <-> Solo <-> ... Solo <-> LED module <-> Bluetooth.
This way the LED module will "see" all commands/responses from other modules and may be also programmed using a separate command set.

voroshkov, 2017-10-09:
Implemented in Chorus-LED-Module project. 
https://github.com/voroshkov/Chorus-LED-Module

closed 2017-10-09 by voroshkov