voroshkov/Chorus-RF-Laptimer #40: [request] - Manual RSSI input

jRreis, 2017-07-09:
Request for manual RSSI input in Text format in android APP.

RSSI can only be Set/Clear and Increased or decreased by the value of 1, every time +/- is pressed. 

It would be a plus to allow RSSI values to be directly inserted, as text format, for manually set the value. 
E.g.. if we want to set manually an RSSI of 243, "+" must be pressed until that values is reached.  In text, we would be able to directly input the value "243".

FranzAr, 2017-07-09:
+1
Great idea, maybe RSSI value could be stored, as you don't really have to change it a lot for different quads.

rcdesigns, 2017-07-15:
Yes, just storing the last RSSI value would be great.  I find I'm always using around 200 and don't need to calibrate every time.

kestux75, 2017-08-22:
+1 for storing the last RSSI

voroshkov, 2018-04-27:
done in release 0.6.1: https://github.com/voroshkov/Chorus-RF-Laptimer/tree/0.6.1

closed 2018-04-27 by voroshkov