voroshkov/Chorus-RF-Laptimer #43: Feature request : allow communication with LapTimer modules using OTG USB device (3DR telemetry receiver, FTDI, ...)

sky022, 2017-07-12:
Hi,

This is the first issue I write on GitHub... :)
I made several laptimer modules since a few days, and it works like a charm. I'm using Bluetooth to UART modules as recommended, but the range is not so great.

I was thinking : in fact, the whole thing is based on serial communication. To extend the range of operation, we could use something else instead of Bluetooth, that would be connected as an OTG USB device on the smartphone.

For instance, I was thinking about using a 3DR radio telemetry combo like this one : https://www.banggood.com/3DR-Radio-Telemetry-433MHZ-Module-For-APM-APM2-Europe-p-85233.html?rmmds=search

Unfortunately, for the moment the LapTimer app only connects to Bluetooth devices. Do you think you could implement something to improve the application, so that any other supported serial communication device connected via OTG USB can be used ? I'm thinking about EZ-GUI app also, which supports this kind of thing.

Thank you for reading :)

BR,
Fabien (sky022)

allsiolek, 2018-05-05:
I'm using 3dr telemetry radio right now and i just found this post. It's working, you have to setup radio properly. I'm recomending to use mavlink protocol, low netID, air speed > 128.  
First radio is connected to last node, second to bluetooth / wifi module. Sometimes i have to connect two times (connect -> disconnect-> concect) with Chorus to proper readings.

voroshkov, 2018-05-07:
@allsiolek, Thanks for the info! 
What's the longest range you were able to achieve with 3dr?
How do you connect it to wifi/bt module? As I can see, one of the modules comes with the USB converter to connect to PC directly. (https://www.banggood.com/3DR-Radio-Telemetry-433MHZ-Module-For-APM-APM2-Europe-p-85233.html)

allsiolek, 2018-05-12:
I didn't try further than 100m thru the buildings with stock antenna. I used this two telemetry radio:
https://www.banggood.com/500mW-3DR-Radio-Telemetry-AirGround-Module-433MHz-915-MHz-For-MWC-APM-PIXHAWK-Pirate-p-982005.html?rmmds=search&cur_warehouse=CN

https://www.banggood.com/3DR-500mW-915MHz433MHz-Radio-Telemetry-Module-w-OTG-for-Android-Phone-p-1140481.html?rmmds=search&cur_warehouse=CN

With first one i have stable connection. Second thing is that radio from second link hae a bug. For correct work you have to cut trace (https://diydrones.com/forum/topics/3dr-telemetry-always-bootloader).

First telmetry radio is connected to Chorus Laptimer thru RS232 TTL, second one is connected also thru Rs232 to  wifi DT06 module. I also tried with bluetooth module (HC-06) but bluetooth+telemetry radio works worse.

I did try use HC-12 433MHz modem but it didn't work. I think beacuse this modem dosen't suport full duplex transmision.

If you will able to add usb connection option it will be simple but now is still fine.
I will test for you range of stable link in open air.

voroshkov, 2018-05-14:
@allsiolek  Thanks a lot for the detailed description!