voroshkov/Chorus-RF-Laptimer #50: Race Report - Ability to add/remove avail info

cbnlnk121, 2017-09-04:
I see that Freq. was added, then removed from the .csv report. Is it possible to make it an option to select what you'd like in report?

Examples to turn on/off:
Lap
pilot name
freq.
band/channel
RSSI (for better tuning)

Thanks in advance.

voroshkov, 2017-09-05:
I think this should not be configurable. Just need to add all these fields to the report

cbnlnk121, 2017-09-05:
@voroshkov That sounds fine to me. Thanks for making and keeping up this project! It's appreciated by more than write here. We used it for the first time (only 2 pilot setup) this past weekend and it was smiles all around.

voroshkov, 2018-04-27:
implemented in release 0.7.1: https://github.com/voroshkov/Chorus-RF-Laptimer/tree/0.7.1

closed 2018-04-27 by voroshkov