voroshkov/Chorus-RF-Laptimer #56: Different Modes for the Android app

ty8447, 2017-10-01:
Hello, I have been using the Android app a lot and love it, but I was wondering if there was a modded version of the app that includes a point mode where there is a countdown timer and during that set time, people race to get the most amount of laps. When the time is over people can finish their lap and then they are done. This is similar to how normal multigp races are scored.

highway11, 2017-10-02:
I've added this feature to my branch. It's fairly crudely implemented but you can do a timed race for 1.5, 2, or 2.5 minutes. It announces time remaining at certain intervals as well as when the time is up

ty8447, 2017-10-04:
@highway11 How would I set that up? Would I have to upload new code to the pro mini and/or download a modded version of the Android app?

highway11, 2017-10-04:
You would just need the app. There is no code I have changed for the Arduinos

closed 2017-10-06 by ty8447