voroshkov/Chorus-RF-Laptimer #59: Cannot get the RF rx5808 module to work

ty8447, 2017-10-05:
I recently ordered 4 rx5808 modules from Banggood and when they finally arrived and I removed the metal casing, the resistor that you explain how to remove is not installed from the beginning. I solder the wires following the diagram and cannot get the app to let me calibrate/change the frequency or pilot. What does the "Lipo Voltage" go into (Red lead on the diagram), and how might I go about fixing this issue?

voroshkov, 2017-10-05:
1. Please follow the guide in the Troubleshooting section to resolve the app issue.
2. The LiPo Voltage wire should go to + of the lipo you're using to power the Chorus

ty8447, 2017-10-05:
Thank You @voroshkov, My Bluetooth baud rate wasn't set.

closed 2017-10-05 by ty8447