voroshkov/Chorus-RF-Laptimer #63: First Pilot/Device only reads 1018 for the RSSI level no matter what

ty8447, 2017-10-07:
I have two lap timers and when they connect with the phone, the lap timer with the Stepdown (Timer 1/ Pilot 1) does not give an accurate RSSI level. The second lap timer on the other hand, does work in the app and pc version.

voroshkov, 2017-10-07:
please describe the sympthoms

ty8447, 2017-10-08:
When I open the app and have two Solo's connected, the second solo will give an accurate reading of the rssi value and the first solo that has the 5v step down displays 1018 for it's value regardless of Band/Channel. When you start the race, as you mentioned in another issue that they will go to zero to help with communication and that happens for both (so the software is controlling that).

voroshkov, 2017-10-08:
Maybe you have bad (or not patched) vrx.
Try measuring actual voltage on its RSSI pin - it should be around 0.5 - 1.5 V

ty8447, 2017-10-10:
If I use a multimeter to measure the rssi pin on the rx5808 using the positive lead for the rssi and ground being anywhere else (preferably the 5v step down) I get a reading of about 5v.

voroshkov, 2017-10-10:
that's wrong RSSI. probably your RX5808 is damaged

closed 2017-10-23 by voroshkov