voroshkov/Chorus-RF-Laptimer #64: Trigger on peak RSSI

kestux75, 2017-10-09:
Trigger lap time on peak RSSI. With only rssi > rssiThreshold, precision is very sensitive to vtx/antenna power and  rssiThreshold value.

voroshkov, 2017-10-09:
Please clarify. Is there an issue? Or it's a suggestion? What's the idea?

kestux75, 2017-10-09:
It's not a big issue,  but I think it would be a an improvement. Sometimes then two or more quadcopters passes finish line at the same time, I have not correct order registered. I think it's because now lap time is triggered as soon as rssi > rssiThreshold. But it's not exactly the closest point to the gate and its very sensitive to rssiThreshold setting. The closest point would be when rssi is at the max value. 
My suggestion is after rssi > rssiThreshold to wait for rssi peak to register lap time.

Sorry for my english.

voroshkov, 2017-10-10:
Got it. The idea is really nice. Need to think and experiment.

elvince, 2017-10-13:
In fact, you can take some inspiration from https://github.com/scottgchin/delta5_race_timer
His program is better suit to get the best lap time because it sample different value one the threshold is reached and wait until the rssi value is lower. Then it "save the highestepeak time as the lap time. This will greatly improve your code.

BTW, thanks a lot for sharing to the community, this is a great stuff.

voroshkov, 2017-10-13:
Thanks for pointing out. I'll take a look.
At the same time I heard rumors that Chorus worked better than Delta5 in certain conditions ;)

elvince, 2017-10-13:
Might be, it would be great to know what were those conditions.
The simple implementation is to read multiple value between Threshold  reached / unreached and keep the highest rssi peak for the laptime.
Then we can see if there are "false positive" during this period and check delta value after/before the peak and use that to compare to the 2nd "best value". If gap are big, the 2nd value might be the best one. What's important to improve the system is to know in real testing, if we have wave rssi peak. I mean, you start getting bigger value, then drop a bit then up again then drop.. up ... and got under the threshold. If this is the case, the simple implementation won't work and we need to create a more robust approach.
 I have no clue how it will react in reality, it is just coding approach.
I need to create my timer first ^^

voroshkov, 2017-10-13:
True. It would be great if someone had enough courage and time to make the research and suggest a good algorithm :)
On the other hand I strongly believe that the "right" construction of the laptimer box might also be the clue to more sharp readings. E.g. narrowing down the "RF sensitivity window" (whatever it might be) will eliminate the necessity of improving the software.

anunique, 2017-10-14:
well this could be possible if the first gate pass == tracker pass.
i woulnt use the way delta5 uses since they so somethink like:
((time of rssi trigger value edge up)+(time of rssi trigger value edge down)) / 2
if the tracker is placed directly infront of the first gate you could simply increase the value till its falling down a special rssi value ?

fape, 2017-10-26:
By me you can get some inspiration from [Chickadee laptimer58](https://www.rcgroups.com/forums/showthread.php?2558061-Chickadee-Tech-DIY-Laptimer) too.