voroshkov/Chorus-RF-Laptimer #65: Ability to change VOLTAGE_DIVIDER_CONSTANT in app instead of hardcoded

Maarten112, 2017-10-12:
Hi Voroshkov,

Let me start by saying: THANK YOU SO MUCH for this great project and making it opensource.

I noticed that in the hardware schematics you use a voltage divider with 10K and 1K, however, I found this a bad choice for a 3S lipo that I will use since the resolution is really low with this kind of voltage divider and the use of a 3S Lipo:

Full 3S lipo: 12.6V --> 1.145V out of the voltage divider
Empty 3S lipo: 9.9V --> 0.9V out of the voltage divider

The resolution with this divider is 0.245 volt (+/- 50 digital resolution)

I choose to use a voltage divider with 10K and 22K, since it gives much better resolution:

Full 3S lipo: 12.6V --> 3.938V out of the voltage divider
Empty 3S lipo: 9.9V --> 3.094V out of the voltage divider

The resolution with this divider is 0.844V (+/- 172 digital resolution)

In the app (AppState.java) on line 43 you hardcoded the voltage divider constant that is used for the final battery voltage calculation:

public static final double VOLTAGE_DIVIDER_CONSTANT = 11; //(10K + 1K)/1K

It would be great to not hard code this, but make it editable in a configuration menu (or maybe advanced configuration menu?).

voroshkov, 2017-10-16:
Is your choice of resistors suitable for lipos from 2s to 6s?
On one hand the feature of Lipo monitoring is supplementary, and doesn't require high resolution.
On the other hand, using higher resistances leak less current and provide better monitoring capabilities.

On one hand I'd avoid changes to schematic as many people already developed their devices.
On the other hand it might be solved by allowing to configure it via UI, as you suggest.
At the same time I'd really avoid adding too much configuration options to UI.

Kind of difficult decision :)

Maarten112, 2017-10-16:
Hi voroshkov,

No this voltage divider is only suitable for up to 3S (and gives the best resolution on 3S). A full 4S lipo would already give 5.25V out on the analog pin and probably destroy the arduino. 

I agree with you that the android app should be as clean and simple as possible, there is no point in making it over complicated. I understand your concerns, however I think there is no real good argument to use a 6S lipo for the Chorus laptimer hardware, since we only need a maximum of 5V (or perhaps 12V for an external fan). It would only put more stress on the dc dc regulators to use 6S.

voroshkov, 2017-10-17:
@Maarten112, the device is quite energy-hungry, especially taking into account possible number of nodes, so this is the main reason to power it with batteries of high voltage and capacitance. I find it absolutely reasonable to power it with 6S or whatever else.
Sorry, but the feature you request is not to be implemented now.

closed 2017-10-17 by voroshkov

chipko, 2017-10-17:
I have used a uBEC with 5.25v out without issue but this isn't advisable :D