voroshkov/Chorus-RF-Laptimer #70: TIME RACE OPTION

ivkostic, 2017-10-30:
Hi! Great features so far! We tried it on the rc track for buggies, and is working superb! Only needed option for time limit for counting the laps/best lap. Can it be added 5 10 15 20 and 30 min of race, for counting the laps? This can be very usefull for rc car training

ivkostic, 2017-10-30:
ops i just saw an other topic with the similar header, this should be posted there.

voroshkov, 2017-10-30:
Wow! Are you using 5.8GHz VTx's just for laps tracking?

ivkostic, 2017-10-30:
yes  and it is perfect! we have a tons of ideas for buggy racing

ivkostic, 2017-10-30:
https://www.youtube.com/watch?v=KWu0FfP6fwM

lazar89nis, 2017-10-30:
Hi Andrey,

I wanted to show them how cool is you Android app for quadcopter racing and they took my device and started racing cars with it  :)

I can implement that part in iOS app but they only have android devices that is why i sent them to you :D

voroshkov, 2017-11-01:
Guys, sorry, but so far I don't have plans to extend the functionality of the Android app in this direction. Feel free to contribute and extend it.

voroshkov, 2018-04-27:
duplicate of #21

closed 2018-04-27 by voroshkov