voroshkov/Chorus-RF-Laptimer #74: Error while compiling

FPV-Rover, 2017-11-26:
I have to use Arduino 1.0.5 because it`s the only version which is running without problems on my GalliumOS, while compiling i always get the following error message:

#################################################################################
/usr/share/arduino/hardware/tools/avr/bin/avr-g++ -c -g -Os -Wall -fno-exceptions -ffunction-sections -fdata-sections -mmcu=atmega328p -DF_CPU=8000000L -MMD -DUSB_VID=null -DUSB_PID=null -DARDUINO=105 -D__PROG_TYPES_COMPAT__ -I/usr/share/arduino/hardware/arduino/cores/arduino -I/usr/share/arduino/hardware/arduino/variants/standard /tmp/build7368876175063967868.tmp/chorus_rf_laptimer.cpp -o /tmp/build7368876175063967868.tmp/chorus_rf_laptimer.cpp.o 
In file included from chorus_rf_laptimer.ino:225:0:
sendSerialHex.h: In function ‘uint8_t send4BitsToSerial(uint8_t, uint8_t)’:
sendSerialHex.h:80:16: error: ‘class HardwareSerial’ has no member named ‘availableForWrite’
     if (Serial.availableForWrite() >= 5) { // e.g. "S1R1\n"
                ^
sendSerialHex.h: In function ‘uint8_t sendByteToSerial(uint8_t, uint8_t)’:
sendSerialHex.h:93:16: error: ‘class HardwareSerial’ has no member named ‘availableForWrite’
     if (Serial.availableForWrite() >= 6) {
                ^
sendSerialHex.h: In function ‘uint8_t sendIntToSerial(uint8_t, uint16_t)’:
sendSerialHex.h:106:16: error: ‘class HardwareSerial’ has no member named ‘availableForWrite’
     if (Serial.availableForWrite() >= 8) {
                ^
sendSerialHex.h: In function ‘uint8_t sendLongToSerial(uint8_t, uint32_t)’:
sendSerialHex.h:119:16: error: ‘class HardwareSerial’ has no member named ‘availableForWrite’
     if (Serial.availableForWrite() >= 12) {
                ^
sendSerialHex.h: In function ‘uint8_t sendLaptimeToSerial(uint8_t, uint8_t, uint32_t)’:
sendSerialHex.h:132:16: error: ‘class HardwareSerial’ has no member named ‘availableForWrite’
     if (Serial.availableForWrite() >= 14) {
                ^
chorus_rf_laptimer.ino: In function ‘void readSerialDataChunk()’:
chorus_rf_laptimer.ino:731:67: error: invalid conversion from ‘uint8_t* {aka unsigned char*}’ to ‘char*’ [-fpermissive]
In file included from /usr/share/arduino/hardware/arduino/cores/arduino/HardwareSerial.h:28:0,
                 from /usr/share/arduino/hardware/arduino/cores/arduino/Arduino.h:193,
                 from chorus_rf_laptimer.ino:51:
/usr/share/arduino/hardware/arduino/cores/arduino/Stream.h:76:10: note: initializing argument 1 of ‘size_t Stream::readBytes(char*, size_t)’
   size_t readBytes( char *buffer, size_t length); // read chars from stream into buffer
          ^
chorus_rf_laptimer.ino: In function ‘void sendProxyDataChunk()’:
chorus_rf_laptimer.ino:788:36: error: ‘class HardwareSerial’ has no member named ‘availableForWrite’
##################################################################################
Can someone help me fixing that ? or is there a workaround ? I goggled a lot but didnt found a working fix for that.

Thanks
Moe

voroshkov, 2017-12-01:
Moe, it looks like the versions of libraries you are using are outdated and don't support functions used in the "chorus_rf_laptimer.ino". You have to use more recent version of libraries (not sure if they are tied to Arduino IDE version)

FPV-Rover, 2017-12-02:
Appreciate for that info :+1: Spend 2 Hours to get the Arduino 1.8.5 to run and now i can compile :-) tomorow i will assemble it.
If somebody has the same issue:

The Problem was a missing Java runtime. I got the following failure message:

Picked up JAVA_TOOL_OPTIONS: 
java.lang.Error: Cannot load com.sun.java.swing.plaf.gtk.GTKLookAndFeel
	at javax.swing.UIManager.initializeDefaultLAF(UIManager.java:1351)
	at javax.swing.UIManager.initialize(UIManager.java:1459)
	at javax.swing.UIManager.maybeInitialize(UIManager.java:1426)
	at javax.swing.UIManager.getLookAndFeel(UIManager.java:492)
	at processing.app.linux.GTKLookAndFeelFixer.installGtkPopupBugWorkaround(GTKLookAndFeelFixer.java:79)
	at processing.app.linux.Platform.setLookAndFeel(Platform.java:41)
	at processing.app.Base.<init>(Base.java:224)
	at processing.app.Base.main(Base.java:144)

It was simply just the missing java rte... but the v8 and 7 doesn`t support the Arduino IDE... so the solution was to install the icedtea-next rte (sudo apt-get install icedtea-next) 

Now it runs like a charm.

Cheers
Moe

closed 2017-12-02 by FPV-Rover