voroshkov/Chorus-RF-Laptimer #80: PCB 4in1

Tomkanovik, 2018-01-20:
Does enyone have PCB 4in1 with efficient use of space? Please send me link for it.

ps915, 2018-02-16:
Like this?
https://www.youtube.com/watch?v=avLRJ5z0UP4

I am working on it. Hopefully released soon. Actually found one little think i must change on the PCB, rest is working

ps915, 2018-02-16:
[image]

Tomkanovik, 2018-02-17:
yes like this it is very nice :)

targetdozer, 2018-03-26:
What is the Status of the 4 in one? I will build up a Chorus for our Team. PCBs are new to me, so I would like to get some ready to order. 
One question to your Layouts (8in1 and 4in1): Why do you mont the Receivers with the covered side up?
voroshkov writes:
Assembly of a Solo device
Correct positioning of RX5808 module against the finish gate area is vital for correct measurements.
I tried using different types of antennas and shields with RX5808 to achieve the best possible accuracy, and finally found that the module itself works as a short-range directional antenna. 
# The non-shielded side of the module is a surface that should face the gate, so consider this fact upon assembling.

ps915, 2018-06-10:
Here you go:
https://github.com/ps915/Chorus-RF-Laptimer-PCB

ALopes90, 2018-09-25:
@ps915 the @targetdozer question is a very valid one, the "The non-shielded side of the module is supposed to face the gate?" I saw many people building one way and other building the other way, can't anyone test using it? I already order everything to build a 4 in 1 and I will make a 3D printable box to use with it and share it, but depending the way it needs to face the box with be different. Thanks

voroshkov, 2018-09-25:
I used both sides (practical use on our local events, which last from morning till evening). Both work fine. 
I guess there may be differences, but someone needs to invent a method of testing to check which is more accurate ;)

ALopes90, 2018-09-25:
Ok Thanks, Good work