voroshkov/Chorus-RF-Laptimer #90: Chorus RF Laptimer v 0.7.6 Minimal Lap Time can not be set in android 7.0

ammarraisafti, 2018-05-29:
# Expected behavior
When a user clicks on the plus (+) or min (-) icon to set the Minimal Lap Time before starting the race, the ability to increase or decrease the Minimal Lap Time should be possible.

# Actual behavior
Minimal Lap Time can not be set, although I have tried to click multiple times on plus (+) or (-) icons.

# How to reproduce
* Launch app. 
* Click on the plus (+) or min (-) icon in the Minimal Lap Time feature section.
* You will find the problem as I have explained, *Minimum Lap Time can not be set.*

<center>Browser: Samsung Galaxy S6 edge
Operating system: Nougat 7.0
App version: 0.7.6 </center>

# Recording Of The Bug
https://youtu.be/__AXU2pW5ms

voroshkov, 2018-05-29:
You should connect to device first, because Min Lap Time is a property that's set on hardware.

ammarraisafti, 2018-05-29:
Thank you very much for the information you provided, my device is having problems, I will try to connect after fixing it.