voroshkov/Chorus-RF-Laptimer #93: come back from connect via Bluetooth page for screen rotation

somrat420, 2018-05-31:
Expected behavior
it should be stay on the Bluetooth page while screen rotation

Actual behavior
in Bluetooth page it comes back for screen rotation from portrait to landscape .

How to reproduce
* download the app
* open it and click on connect via Bluetooth
* then rotated your phone from portrait to landscape
* note the bug

https://m.youtube.com/watch?v=7-FKICSwnpY

gpolic, 2018-06-28:
I think that there are generally some issues with the Lifecycle and the layouts

To @voroshkov 
Regarding the Bluetooth starting and stopping.
Try to experiment and move it to onStart and onStop (you need to override these methods)
Remove code from onDestroy. The application maybe in memory for hours before the onDestroy is called to clean up memory.

Regarding the layout. 
If the layout is not loaded correctly, then the viewPager maybe null
Check for null first, before doing any actions on it on MainActivity.
(can your viewPager be invisible during a rotation to Landscape ?)

Having said that... I do not have the hardware, so I cannot run or do any tests

voroshkov, 2018-06-28:
@gpolic, thanks a lot!!!
Can you elaborate a bit?
- onStart/onStop/onDestroy - do you mean main activity methods?
- where should I put the code from onDestroy?
- rotation to landscape is not possible 
- why layout may be not loaded correctly?

gpolic, 2018-06-29:
Right, I mean the MainActivity methods
Try to connect Bluetooth on onStart and disconnect it on onStop
The code from onDestroy can go to OnStop
onStart is already there. You can move the initBluetooth that setup the listeners to onStart also.
This is just a test to check the behavior, in terms of Lifecycle

I have checked your manifest and it looks like orientation is locked.
However the Activity DeviceList on the BlueTooth library can be rotated.
I think this is why the user mentions rotation from portrait to landscape
A quick fix, may be, to lock this activity on Portrait orientation. I would try that first, before anything else.

Try to check if viewPager is null, right after this line in onCreate/MainActivity:
mViewPager = (ViewPager) findViewById(R.id.container);

if it is not null, then use it.
If it is null then do something else finish() or Log it, to view it on your logs.

Also on this line in onCreate/MainActivity:
SectionsPagerAdapter mSectionsPagerAdapter = new SectionsPagerAdapter(getSupportFragmentManager(), getResources());

Most of the examples of PagerAdapter I have read are using the constructor without resources. Just the FragmentManager. I am suprised you have getResources.
Try to use the constructor with out getResources.

If the app has been put to the background and then started again, this getResources maybe pointing to a null object or something else.

I am just putting out ideas and trying to help
As said i dont have the hardware so I cannot test anything.

voroshkov, 2018-06-29:
Great! Thanks for valuable input. Will try that stuff!
As for moving BT disconnection to onStop - I tried that yesterday and as a result it stopped connecting after first disconnection. Had not much time to investigate why, so just left that as is.

gpolic, 2018-06-29:
ok sure !  maybe forget about that onStart & onStop stuff.

Another thing for the MainActivity:

    public void onActivityResult(int requestCode, int resultCode, Intent data) {
if (requestCode == BluetoothState.REQUEST_CONNECT_DEVICE) {
e.t.c.

You can insert `super.onActivityResult(requestCode, resultCode, data); ` as the first line of code in this method. Maybe it has to do some clean up when the Activity returns (for example DeviceList activity)