voroshkov/Chorus-RF-Laptimer #97: Temperature Problem

ALopes90, 2018-09-19:
Hi mates, here in Portugal we have a long Summer with high temperatures so many times we have problems with the receivers over heating, I know nothing about arduino, so I ask to you, it would be very difficult to implement a temperature sensor with a fan control  like what this does?

https://www.aliexpress.com/item/WS16-DC-12V-4A-PWM-PC-CPU-Fan-Temperature-Control-Regulator-Thermostat-Speed-Controller-Switch-High/32818888635.html?spm=2114.search0104.3.22.518e6985OFLcWb&ws_ab_test=searchweb0_0,searchweb201602_4_10065_10068_10059_10884_10887_10696_100031_10084_10083_10103_10618_10304_10307_10820_10821_10301,searchweb201603_1,ppcSwitch_2&algo_expid=9b463102-5c70-4326-b069-227134ca6145-3&algo_pvid=9b463102-5c70-4326-b069-227134ca6145&transAbTest=ae803_2&priceBeautifyAB=0

Thanks for all your work, this is just one ideia that I think will help many people, I'm going to make a box for the 4in1 project to use a fan and a controller like the one in the link.

voroshkov, 2018-09-24:
Hi @ALopes90. I think that the fan control is a kind of excess. Instead you may just run your fan at full speed and have no worries about implementing/tuning the fan control.

ALopes90, 2018-09-24:
Thanks, I can't help in that part, I don't know how.
Keep up the great work you all are doing.