voroshkov/Chorus-RF-Laptimer #100: [Feature Request] Serial Enhancement more API Messages (Race End, Pilot finished, target laps... etc.))

El-Tonno, 2019-02-24:
Hi,

i added an esp8266 which is watching the serial output on the Chorus and publish the messages into a Wifi-Mesh. So other devices such as traffic lights, LEDs etc. in the mesh can react independently on the serial output .

Unfortunately some messages are missing. Would be nice if the serial output could be enhanced by the following states: 
- amount of Laps have to be done ( at race start.. before R*R1 for e.g.) 
- information pilot has finished the last lap (after S0L5XXXXXX)
- race finished (all pilots have finished the last lap)
- direct feedback if the amount of laps is changed in the app (like the feedback ignore 1st lap R*11/R*10)

Thanks a lot 
Greets el-tonno

voroshkov, 2019-02-25:
Hi,

Unfortunately these are not the things to be managed by Chorus device itself. 
These parameters are expected to be managed by the race control software, so you need to integrate with the latter to monitor the information you want.
Chorus API just forwards the unknown commands through the chain, so any command not recognized by  Chorus will be forwarded to your ESP.

El-Tonno, 2019-02-25:
I see ... ok this was my little missunderstanding of the Chorus and the Chorus App architecture. I confess i`m not very good in apk developement. 
Now i had a look at the app src and i think RaceResultFragment.java would be right place.

Line 46 : AppState.getInstance().sendBtCommand("R*R1");

Could you please add just one line ? 

something like that-->
  AppState.getInstance().sendBtCommand("LTG*" + String.format("%01X", LAPS_TO_GO) );

With the amount of laps i would be able to calculate the rest of the messages i mentoined above.

Thank you very much
el-tonno

El-Tonno, 2019-03-02:
Hi again ... i please you to help me :-)
Don´t wanna recompile your app and building a derivate. This little additiononal message won´t hurt anybody ... and could be usefull in other projects.
Please just implement this in your next version of the app.
What do you think ?

voroshkov, 2019-03-05:
Hi. Generally it's ok to do. But I'm out of time at the moment. Need about a week to return to this.

El-Tonno, 2019-03-15:
Hi Thanks a lot ...would be nice if this is available in the next Version.

Meanwhile i made the changes myself ...

I added follwing line  47 at RaceResultFragment
AppState.getInstance().sendBtCommand("LTG*" + String.format("%01X", RaceState.lapsToGo));

and made the the lapsToGo static for access
public static int lapsToGo;

So no hurry :-)