voroshkov/Chorus-RF-Laptimer #101: The problem with the application iOS.

Kostya989, 2019-03-17:
I did not find how to contact the author of the application.
There are problems connecting via Wi-Fi. The connection is unstable, it seems that the smartphone is falling asleep and the connection is broken.
And in case of a successful connection, a message appears.
[image]
[image]
Tested on
iPhone 7 iOS 12.1.4
iPad 10.5 iOS 12.1.4

voroshkov, 2019-03-18:
Please check in the Facebook group. If I remember correctly Lazar Djordjevic mentioned that new version of iOS app is going to be released soon. Meanwhile you may change the API Version in Arduino code to 4 to have your app working

Kostya989, 2019-03-18:
Ok, thanks. I will try to figure out where to change the API.

closed 2019-07-03 by Kostya989