voroshkov/Chorus-RF-Laptimer #112: Connecting to chorus over WiFi attempts to connect to 0.0.0.0

StarrrLiteNL, 2020-01-06:
Whenever I try to connect to my chorus with my phone over WiFi, I get the message "connected to 0.0.0.0" and I get the "disconnected" message immediately after that.

I am connected to the correct WiFi network, and I can reach the gateway up (192.168.4.1). All WiFi module settings are correct.

It has worked before on this phone and I can connect without issues from my Nexus 7 tablet.

I suspect the recent software update to Android 10 on my phone broke it.

It always used to be a bit finicky to connect, I always had to connect then wait for the "stay connected anyway" message to come and confirm. After confirmation it would work fine. I can't get that confirmation message to appear anymore tho (I have not seen it ever since the update) perhaps they removed it in Android 10 or the latest OxygenOS?

I have tried turning off mobile data and Bluetooth, but both didn't solve the issue.

Switching the WiFi module to a Bluetooth module does allow me to use the chorus fine, but the WiFi module has a much better range so I would love to use that instead.

Specs:
Phone; (can't connect via WiFi)
Oneplus 6
Android 10
OxygenOS 10.3.0

Tablet; (can connect via WiFi)
Nexus 7 (2013 version)
Android 8
AOSP custom ROM

realhuno, 2020-02-24:
Hi, i have the same issue.  Nobody Can help us?? 
Huawei P30 Pro. Not working... 
Old Sony Phone working...

voroshkov, 2020-02-25:
It looks like the problem is with Android 10 devices only. I don't have such a device to debug so far.

marcfon, 2020-04-29:
Exactly the same issue here.

PR #113 (2020-05-04): Fix connection on android 10
wifiInfo.getNetworkId() seems to require the location permission on Android 10.
This fix uses the ConnectivityManager to avoid adding a new permission.

Tested on Android 10 and 9.

Fixes #112